3cky/mbusd #1: Windows port

gbisheimer, 2015-07-03:
Hi. I'm very interested in your project as I was about to code the exact same thing but for windows.

I readed (here)[http://mbus.sourceforge.net/] that it's possible to run this app under Cygwin. Could you please orient me how to do that? I need to access a couple of Modbus/RTU devices from the net and found this to be the best choice to do that.

Thanks for sharing!

3cky, 2015-07-03:
Hi, hmm, I have no Windows machine for test right now, but first of all Cygwin environment with development tools should be installed as described in https://cygwin.com/install.html. Check packages `make` and `gcc-core` in  **Devel** category are selected, then install all resolved dependencies. After this you could try to compile mbusd using commands:
~~~bash
./configure
make
~~~
In case of errors, post log here, I'll have a look.

gbisheimer, 2015-07-03:
Hi Victor,

Will try that and let you know.

Thanks!

*Ing. Guillermo Bisheimer*

*B&S Sistemas de Control y Equipamientos*

Av. de los Constituyentes 1172

(E3116CIX) Crespo, Entre Ríos

Tel/Fax: (0343) 4950289

Cel: (0343) 154679052

WEB: www.bys-control.com.ar

e-mail: gbisheimer@bys-control.com.ar

skype: guillermo.bisheimer

gbisheimer, 2015-07-03:
We succesfully installed mbusd on Cygwin. We also had to install automake and autoconf packages to make it work.

Everything works great using 8N1 format but we cannot make it work using 8E1 or 8O1 formats. It gives some strange errors on the modbus TCP client (Negative Ack something...)

Did you test it using this formats? We are testing using a real PLC connected to the serial port line COM1. Cygwin is installed in the same computer with Win7x64 OS. We are using Modbus poll client to test the communication to the PLC.

Thanks!

3cky, 2015-07-04:
Yes, parity enabled modes were tested by user who requested this feature, but on Linux. Probably Cygwin have some quirks in parity mode handling, I'll try to check at Monday.

gbisheimer, 2015-07-06:
I've just testes under linux and works ok using 8E1 format. I'll use a VM with linux instead of Cygwin for this application.

Thank you very much for your help and for sharing this project!!!
Works great.

closed 2015-07-06 by gbisheimer