3cky/mbusd #3: RE: mbus-0.2.1

Tigr7, 2015-12-16:
Спасибо большое за оперативный ответ!

Собрал новую версию. С настройками связи тут все нормально, так как чтение с регистров идет на УРА!
Но вот с записью та же проблема - после попытки изменить регистр 6 функцией вылетает ошибка, а демон зацикливается.
К сожалению, я забыл свои учетные данные на гитхаб на работе. Прикладываю файл письмом.

Запускал так: sudo mbusd -p /dev/ttyUSB0 -s 9600 -P 502 -d -v9

[mbus.txt](https://github.com/3cky/mbusd/files/64393/mbus.txt)

3cky, 2015-12-17:
Поправил алгоритм чтения ответов функции 6 и подобных, попробуйте скачать и собрать мастер-версию https://github.com/3cky/mbusd/archive/master.zip

closed 2015-12-17 by 3cky

Tigr7, 2015-12-18:
Спасибо огромное!! Работает отлично и стабильно!!! Поставил на 12 часов. Больше миллиона запросов и записей - ни одной ошибки! Очень круто!! Спасибо!!!!!
Можно мне давать ссылку на эту версию на форумах по соответствующей тематике?

3cky, 2015-12-18:
Отлично, я поднял версию mbusd до 0.2.2, она доступна в releases, и ссылку лучше давать именно на нее: https://github.com/3cky/mbusd/releases/tag/v0.2.2

commit 85943e89 (2015-12-17): Fix #3