3cky/mbusd #4: -m option not documented in mbusd's man page

mbs38, 2015-12-26:
Hi,
I was looking for a configuration option for serial parity bits and only found it in the source code. Maybe it should be added to the man page as well.

Best regards & thanks for the good work

3cky, 2015-12-28:
Hmm, but `-m` option description should be already present in man page: https://github.com/3cky/mbusd/blob/master/doc/mbusd.8.in#L52-L53

Also it should be described in command help (`-h`). Could you check that mbus is compiled using latest sources?

mbs38, 2015-12-28:
Yep. I accidently used an older version that was still sitting in my download directory. Sorry about that.
Anyway, thanks for the quick reply :)

3cky, 2015-12-29:
OK, no problem! :)

closed 2015-12-29 by 3cky