3cky/mbusd #6: General starting problem

rwalli, 2016-02-07:
Sorry for posting here an issue but i did not found any discussion-forums.

I tried to get mbusd working on arm-linux (mbusd -s 19200 -d -p /dev/ttyAMA0 -v 9 -m 8N1) and get the following error:

```
1454870063:339873 conn_open(): accepting connection from 10.0.0.222
1454870063:340152 queue_new_elem(): length now is 1
1454870063:340529 conn[10.0.0.222]: state now is CONN_HEADER
1454870063:340804 conn_loop(): select(): max_sd = 5, t_out = 000060:000000 
1454870063:342242 conn_loop(): select() returns 1 
1454870063:342785 conn[10.0.0.222]: state now is CONN_RQST
1454870063:343036 conn[10.0.0.222]: state now is CONN_TTY
1454870063:343256 tty: state now is TTY_RQST
1454870063:344516 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
1454870063:344862 conn_loop(): select() returns 1 
1454870063:351361 tty: written 8 bytes
1454870063:352617 tty: request written (total 8 bytes)
1454870063:353804 tty: state now is TTY_RESP
1454870063:354926 tty: estimated 7 bytes, waiting 57811 used
1454870063:356035 conn_loop(): select(): max_sd = 4, t_out = 000000:057811 
1454870063:415078 conn_loop(): select() returns 0 
1454870063:416256 tty: response timeout
```

To ensure my modbus device is working, I wrote a short test program with libmodbus (modus_read_registers) which work as expected.

Can you please point me to the right direction where to start?

Robert

hammondb4, 2016-02-14:
Have you checked your device answering speed? Maybe you have to change your wait time if it's a slow device, with -W parameter....
Also I use QModMaster or CAS Modbus Scanner for trying my modbus setups....both of them, specially CAS, tell you the type of Modbus error....

rwalli, 2016-02-17:
Thank you very much!
It was the -W  parameter.

closed 2016-02-17 by rwalli