3cky/mbusd #7: Problems changing parity

hammondb4, 2016-02-14:
I have mbusd setups working well with default 8N1 option.
When I try to config with Odd (-m 8O1) or Even (-m 8E1) parity, it's not working properly.
It seems I always can communicate with my devices with 8N1 setup no matter what '-m' option I choose, but never been able to talk with 8E1 or 8O1 devices.

3cky, 2016-02-15:
Hmm, could you provide some details about  your environment? OS, hardware, etc.

hammondb4, 2016-02-15:
I have differents setups...
A couple x86 machines and another Raspberry2, both running Openmediavault, wich is debian (Kernels 4.4, 3.16 & 4.04). Tryied with CH340 and PL-2303HX usb to 485 adapters.
In all setups comms are ok with 8N1 devices. When I try -m 8O1 or -m 8N2 to access other devices, is not working; instead (no mather -m option) always could stablish comm with 8N1 devices....

3cky, 2016-02-15:
Are you sure RTU endpoints are configured properly for 8O1/8N2 comms? At least 8E1 should work, as reported by other users.

hammondb4, 2016-02-15:
I tryied it with the adaptor in windows using qmodbusmaster and I can communicate with each device with it's config, so it seems the endpoints are correctly configured.
Will do more tests today.

hammondb4, 2016-02-15:
I tested again with some adapters and mode always seems to be set in 8N1. I tested a device with integrated bridge (TCP2RTU) and it works as spected when configuring parity and stop bits. No luck with mbusd.

3cky, 2016-02-16:
No, there is no specific position for -m option. BTW, mode string is checked for validity, so you could provide some invalid mode string (like -m 8N3) to check option is recognized at least.

hammondb4, 2016-02-18:
Yes, I tried with invalid strings and the checking is working ok.....
But after '-m 8N2' or '-m 8E1' seems only talking with 8N1 endpoints, not 8N2 nor 8E1....

3cky, 2016-02-22:
@hammondb4 made some fixes, please test tty mode setup using mbusd master version and PL2303-based adapter (_not_ CH340 because it seems its driver still lacks support of parity/stop bits, see https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ch341.c#L363).

hammondb4, 2016-02-28:
@3cky Sorry for the delay. I'll be trying it this monday with PL2303 adapter. Will redownload master src and recompile it again and report it.

hammondb4, 2016-03-01:
@3cky Ok. Tested it with PL2303 adapter and 8N2 mode and now it's working fine. Seems the problem it's solved using this adapter. Thanks @3cky 
Also, could someone recommend one nice usb adapter? Tried CH340 and PL2303 and both seems similar, despite the mode issue. Any other options?

3cky, 2016-03-02:
@hammondb4 great, thanks for testing! So I close this issue and will release 0.2.3 within the next couple days.
About adapters - personally I tend to avoid using CH340 in production, just because it also known as "the worst usb serial chip in the world". :) PL2303 performs much better, if it's not poor chinese fake.
Main tunable config value is response wait time, it seems its default value (50 ms) is too small in many cases, so 500-1000 ms is more suitable.

closed 2016-03-02 by 3cky