3cky/mbusd #8: Can't run Make, make script links to specific automake version

Fohdeesha, 2016-08-22:
Trying to do a clean install of mbusd (cloned from git), once I get to the make step I run into the error below. This is on a clean install of the latest stable Debian (8) amd64, who's latest stable automake is 1.14.1.

It is my understanding that make scripts shouldn't reference specific versions at all, for example: 
http://stackoverflow.com/a/37669184

This is the output I get when I run make on a system with the latest debian stable automake installed, 1.14

root@modbus:~/mbusd.git# make
 cd . && /bin/bash /root/mbusd.git/missing automake-1.15 --foreign
/root/mbusd.git/missing: line 81: automake-1.15: command not found
WARNING: 'automake-1.15' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:378: recipe for target 'Makefile.in' failed
make: *** [Makefile.in] Error 1
root@modbus:~/mbusd.git#

Fohdeesha, 2016-08-22:
I followed the steps here from anotherp roject with the same issue https://github.com/Jasig/mod_auth_cas/issues/97

I ran "autoreconf -f -i" and then I was able to fully ./configure, make, and make install, and now it is working as expected. It seems it's related to cloning from git as timestamps are not retained and it triggers a rebuild

nickma82, 2016-12-12:
Checked out the repo today (master branch), with the same problem on a RaspberryPI.
Automake version 1.11.6

`autoreconf -vfi` fixed it too.

nickma82, 2017-11-03:
Shouldn't all the build aggregates (i.e. _Makefile.in aclocal.m4 configure doc/Makefile.in libtool src/Makefile.in_) not being checked in?

S.t. in order to compile everyone would have to issue a standard `./configure` followed by a 'make'

3cky, 2017-11-14:
Really I already thinking about transition to cmake for a long time. Will try to find some time to make it done. Also created #22 to track work progress.

nickma82, 2018-05-16:
close it plz

closed 2018-05-16 by Fohdeesha