3cky/mbusd #14: Concurrency issues?

dvader, 2017-02-17:
I used to get crashes when logging "conn_open(): error in accept()", in particular, the call strerror(errno) crashes. After some reading it seems strerr is not re-entrant safe http://stackoverflow.com/questions/900338/why-cant-i-use-strerror. I am not sure if re-entrancy is the cause of the crash but after removing the function call and logging only errno, the software doesn't crash anymore.

However, it still sometimes get the error:
1485184829:392664 conn_open(): error in accept() (11)
1485184829:413938 tty: error in read() (11)
and sometimes only the accept error without the tty error.
I'll get the error more often if I use two clients to make requests.

I wonder if this error is caused by re-entrancy issues? Maybe there is a need to put mutexes on some function calls to make them thread safe?

dvader, 2017-02-17:
I also recently got the following error:
` 1485184870:634270 conn[192.168.2.51]: state CONN_TTY deadlock.`

3cky, 2017-02-18:
Hmm, thanks for reporting. Since mbusd is single-threaded application, re-entrancy shouldn't be an issue. I will look into it, but could you provide some context? What OS and hardware are you using (especially tty), also some info about client app and typical requests pattern.

dvader, 2017-02-22:
Linux 2.4.21-rmk1 #1 Mon Mar 5 15:00:39 CET 2012 armv4l unknown
Cirrus Logic ARM EP7312-CBZ http://ca.mouser.com/ProductDetail/Cirrus-Logic/EP7312-CBZ/?qs=bUPhaerQQeGSZ7MA5Dwd%2FA%3D%3D
I am not using RxTx control, it's a full duplex UART.
I am using qModmaster https://sourceforge.net/projects/qmodmaster/ as a client and I am running two of them with 123 register request every second.
Here's a sample request and response:
`INFO 2017-02-17T16:33:33.685 ..\qModMaster\src\modbusadapter.cpp @ 379 Rx Data :  "26  73  00  00  00  f9  01  03  f6  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  0d  00  02  00  00  00  00  00  02  00  11  00  11  00  10  00  22  00  18  00  c0  00  a8  00  02  00  3b  00  ff  00  ff  00  00  00  00  00  c0  00  a8  00  02  00  33  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  " `
` INFO 2017-02-17T16:33:33.685 ..\qModMaster\src\modbusadapter.cpp @ 226 Modbus Read Data return value =  123 `
` INFO 2017-02-17T16:33:33.685 ..\qModMaster\src\mainwindow.cpp @ 543 Packets sent / received =  9843 , errors =  4 `
` INFO 2017-02-17T16:33:34.574 ..\qModMaster\src\modbusadapter.cpp @ 161 Modbus Transaction. Function Code =  3 `
` INFO 2017-02-17T16:33:34.574 ..\qModMaster\src\modbusadapter.cpp @ 190 Modbus Read Data  `
` INFO 2017-02-17T16:33:34.574 ..\qModMaster\src\modbusadapter.cpp @ 360 Tx Data :  "26  74  00  00  00  06  01  03  13  ec  00  7b  " `
` INFO 2017-02-17T16:33:34.761 ..\qModMaster\src\modbusadapter.cpp @ 379 Rx Data :  "26  74  00  00  00  f9  01  03  f6  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  0d  00  02  00  00  00  00  00  02  00  11  00  11  00  10  00  22  00  19  00  c0  00  a8  00  02  00  3b  00  ff  00  ff  00  00  00  00  00  c0  00  a8  00  02  00  33  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  " `
` INFO 2017-02-17T16:33:34.761 ..\qModMaster\src\modbusadapter.cpp @ 226 Modbus Read Data return value =  123 `
` INFO 2017-02-17T16:33:34.761 ..\qModMaster\src\mainwindow.cpp @ 543 Packets sent / received =  9844 , errors =  4 `

nickma82, 2017-11-03:
@dvader any updates on that?

nickma82, 2018-06-18:
@3cky plz close that ticket

closed 2018-06-19 by 3cky