3cky/mbusd #18: RTU/ASCII protocol selection and Modbus Master Mode questions

shaotangyu, 2017-04-28:
Hi,

We have checked Advantech EKI-122x Modbus gateway device and have two questions want to raise.
1. The mbusd does not have RTU/ASCII protocol selection in parameter list. Why users do not need to specify the RTU/ASCII mode for mbusd ? 

2. There are two kinds of Modbus gateway as below. It seems mbusd support slave mode, will it support master mode in the future? 

(1) master mode: translate RTU/ASCII to TCP
[SCADA Server]---serial---[DUT]----ethernet---[PLC]

(1) slave mode: translate TCP to RTU/ASCII
[PLC]---serial---[DUT]----ethernet---[SCADA Server]

Thanks for your advice. 

Regards,
Steven Yu

3cky, 2017-05-03:
Sorry for late response. As mbusd is supposed to be a network gateway to Modbus RTU devices, neither ASCII protocol nor TCP master mode are currently supported. This could be changed in the future, but timeline for implementing such functionality currently is not defined.

nickma82, 2017-11-03:
For ASCII implementation please wait for response at https://github.com/digitizer0/mbusd/commit/ee0b3601bb4402f82058f215d695993794aa7bb9#commitcomment-25389020

closed 2019-08-07 by stale[bot]