3cky/mbusd #21: Regarding Converting 4 Modbus RTU to Ethernet

abhishekricky, 2017-11-14:
Hi,

Is it possible to have 4 usb ports converted to Ethernet on different ports via this software ?

3cky, 2017-11-23:
Yes, it's possible, but you should run mbusd instances using custom script because current systemd service doesn't support multiple TCP port configurations.

3cky, 2017-11-29:
I've added support for multiple mbusd instances in 3ea2084.

nickma82, 2017-12-01:
@abhishekricky please close the ticket

nickma82, 2018-01-20:
Consider two ModbusRTU (Serial) devices with the same address on two different serial Ports and a request sent from TCP to be sent via RTU. 
How would one decide which RTU-serial port to prefer? If one sends it out on both serial buses, which response to choose?

IMHO nothing like that is integrated atm.
Best

joeychen, 2018-01-20:
In general, a routing table can be achieved for gateways with two or more serial ports.

Scenario 1:
If the Modbus devices with different slave IDs are connected to different serial ports on a gateway,
slave ID can be used to route to different serial port.
For example, the Modbus requests with slave ID 001~005 will be routed to serial port1; the Modbus requests with slave ID 006~010 will be routed to serial port2.

Scenario 2:
It's the scenario that you mentioned, two Modbus device with the same slave ID are connecting to different serial ports.
For example, there are a device A and a device B with the same slave ID 1. 
The device A is connected to serial port 1 and device B is connected to serial port 2. 
If gateway has a virtual-to-real slave ID mechanism, 
on the Modbus Master side, the Modbus request for device A is recognized by slave ID 1 
and the request for device B is recognized by slave ID 2. 
When gateway received a command with slave ID 2, it can just minus 1, and send this command with slave ID 1 to serial port 2.

3cky, 2018-01-22:
> Is it possible to support multiple serial port in one instance with one TCP port?

Since I trying to keep mbusd functionality as simple as it could be, adding multiple serial ports support with full-fledged Modbus routing table isn't something that looks good to me. Probably the best (Unix-)way to achieve this functionality is a dedicated app implementing Modbus TCP routing atop of multiple mbusd instances.

joeychen, 2018-01-23:
Has such app been already implemented by anyone in github ?
If not, I was wondering if you have more detailed information for implementation?
For example, since mbusd is binding to a socket, 
how to use another app to route packets atop of mbusd? 
Is there a way to achieve this function without modify mbusd?

nickma82, 2018-01-23:
Side question: Why not using two TCP front-ends (i.e. two ports) and route on client side?

nickma82, 2018-01-23:
> how to use another app to route packets atop of mbusd?
> Is there a way to achieve this function without modify mbusd?

I'd give one of the [various python modbus libs](https://stackoverflow.com/questions/17081442/python-modbus-library) a go.
If you have static routing tables (i.e. your modbus devices do not change) it shoudn't be that complicated. Two to four man days?

nickma82, 2019-05-26:
@abhishekricky Concerning your original question: This is now possible since mbusd supports [setup via systemd](https://github.com/3cky/mbusd#systemd).

Please close this ticket

closed 2019-10-23 by stale[bot]