3cky/mbusd #26: cmake is heavyweight

kb0thn, 2018-01-05:
Running the latest version of cmake is a heavy requirement to build in place on embedded and older systems. Building it really just comes down to:

```
gcc *.c -DPACKAGE=\"mbusd\" -DVERSION=\"fromGIT20180104\" -DLOG -DTRXCTL -DDEBUG -o mbusd
```

kb0thn, 2018-01-22:
A barebones makefile would be great. On 4 of the last 5 targets I built it for, getting cmake and infrastructure going wasn't possible. Likely m4 and autotools wouldn't have been either. But the bare metal gcc above buildt it fine.

nickma82, 2018-05-16:
> A barebones makefile would be great.

Do you have a pull request?

Alternatively add a  gitlab-ci target and let it compile there, [like that](https://github.com/3cky/mbusd/blob/master/.gitlab-ci.yml#L44)

vanrein, 2019-07-13:
@kb0thn 
CMake can also cross-compile, of course, but you would need a suitable toolchain.  But a much simpler option should work as well with Debian multiarch:

 1. Have qemu started for foreign binaries
 2. Add your architecture to dpkg
 3. Install gcc:<arch> and binutils:<arch> but keep the CMake for the default platform
 4. cmake and make
 5. Move the software to your target platform

We've been doing these things in a Docker, to avoid removing the platform's own gcc and binutils.  Docker may even be able to run your foreign binaries since it has the qemu support builtin, at least on some host platforms.

We did find that some Debian packages are platform-neutral, but then continue to depend a platform-specific form, about which it then got confused.  So in reality, your mileage would vary, but this is so simple you should give it a try.

closed 2019-12-10 by stale[bot]