3cky/mbusd #27: Service Fails to start raspberry pi

sgel83, 2018-02-18:
If I run the command manually it starts with the specified configuration file no problem.  If I try to run it as a service it fails.

`root@raspberrypi:~# systemctl status mbusd@ttyUSB0
● mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled)
   Active: failed (Result: start-limit) since Sun 2018-02-18 21:01:15 UTC; 36s ago
  Process: 675 ExecStart=/bin/mbusd -d -v2 -L - -c /etc/mbusd/mbusd-%i.conf -p /dev/%i (code=exited, status=203/EXEC)
 Main PID: 675 (code=exited, status=203/EXEC)

Feb 18 21:01:15 raspberrypi systemd[1]: mbusd@ttyUSB0.service holdoff time over, scheduling restart.
Feb 18 21:01:15 raspberrypi systemd[1]: Stopping Modbus TCP to Modbus RTU (RS-232/485) gateway....
Feb 18 21:01:15 raspberrypi systemd[1]: Starting Modbus TCP to Modbus RTU (RS-232/485) gateway....
Feb 18 21:01:15 raspberrypi systemd[1]: mbusd@ttyUSB0.service start request repeated too quickly, refusing to start.
Feb 18 21:01:15 raspberrypi systemd[1]: Failed to start Modbus TCP to Modbus RTU (RS-232/485) gateway..
Feb 18 21:01:15 raspberrypi systemd[1]: Unit mbusd@ttyUSB0.service entered failed state.
root@raspberrypi:~#
root@raspberrypi:~#
root@raspberrypi:~#
root@raspberrypi:~# mbusd -d -v2 -L - -c /etc/mbusd/mbusd-ttyUSB0.conf -p /dev/ttyUSB0
18 Feb 2018 21:02:09 mbusd-0.3.0 started...
`

3cky, 2018-02-19:
Hmm, that's weird. It's the Raspbian Linux, right? Which version/release?

sgel83, 2018-02-20:
root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux

3cky, 2018-02-20:
Ok, please also show the output of the `cat /etc/os-release` command

sgel83, 2018-02-20:
as requested:

root@raspberrypi:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

3cky, 2018-02-20:
Thanks for the report! I've commited fix for this bug, please rebuild the mbusd from latest sources using updated [installation instructions](https://github.com/3cky/mbusd#installation-instructions) and please let me know if that fixed the issue.

sgel83, 2018-02-20:
Worked perfect, thanks.

root@raspberrypi:~/mbusd/mbusd.git/build# systemctl status mbusd@ttyUSB0
● mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled)
   Active: active (running) since Tue 2018-02-20 21:45:10 UTC; 7s ago
 Main PID: 10561 (mbusd)
   CGroup: /system.slice/system-mbusd.slice/mbusd@ttyUSB0.service
           └─10561 /usr/bin/mbusd -d -v2 -L - -c /etc/mbusd/mbusd-ttyUSB0.conf -p /dev/ttyUSB0

Feb 20 21:45:10 raspberrypi systemd[1]: Started Modbus TCP to Modbus RTU (RS-232/485) gateway..
Feb 20 21:45:10 raspberrypi mbusd[10561]: 20 Feb 2018 21:45:10 mbusd-0.3.1 started...

3cky, 2018-02-21:
Great! I will close this issue.

closed 2018-02-21 by 3cky