3cky/mbusd #28: Connecting Modbus RTU master to Modbus TCP slave

mgalemin, 2018-03-16:
Hi,

Could you please let me know if I can use your gateway for connecting a Modbus RTU master (RS-485 interface) to a Modbus TCP slave (Ethernet interface, acts as a TCP server)? For example, can mbusd work as a TCP client and connect to a remote TCP server?

Thanks,
Maks.

nickma82, 2018-06-18:
Hi @mgalemin 
mbusd accesses a modbus RTU instance (i.e. reading its values) and provides a Modbus TCP server (serving the read out values).

> For example, can mbusd work as a TCP client and connect to a remote TCP server?
I think that is exactly the other way around - right?

deneskelenyi, 2018-07-09:
NO, this won't work. You would need something that creates a map to begin with.
RTU master wants so send something to NODE 1. So it emits an RS-485 packet, addressing NODE 1.

Your gateways should know to listen to NODE 1, then it should know what TCP address NODE 1 is on.

You would need something like a translation map:

Node1, 192.168.11.1:502
Node2, 192.168.11.2:502

No such mapping exists in mbusd.  The good news is, that you can put something together like this pretty easily using a library. I achieved a TCP:RTU mapping via j2mod (Java) where the Java host (Raspberry PI) is acting as an RTU slave for (near) real time sensors and acting as a TCP master for "slower" data requests (e.g. what's the temperature in the toilet). 

Funny thing is that I use mbusd with this too, so the TCP request goes back to a serial link. The rationale in this, is that I can kill a master and start a developer master without pluggin 2-3 serial dongles over to an other host on the other side of a house.

closed 2019-08-07 by stale[bot]