3cky/mbusd #30: Logfile in case of systemd service

zmata, 2018-03-28:
Hi, what is logfile in case of MBUSD systemd service?
Is it posible to specify it in conf file?
Thanks

sgel83, 2018-04-04:
By default:

/var/log/mbus.log

This is staticly defined in the log.h file.
https://github.com/3cky/mbusd/blob/8f2ccf1ec092cdc029d2fa0109b6a3363f086f5f/src/log.h#L42-L43

Currently its not possible to set it in the configuration file.

sgel83, 2018-04-04:
What you can do though is change the ExecStart= line in the systemd

https://github.com/3cky/mbusd/blob/8f2ccf1ec092cdc029d2fa0109b6a3363f086f5f/systemd-units/mbusd%40.service.in#L6

ExecStart=@CMAKE_INSTALL_FULL_BINDIR@/mbusd -d -v2 **-L -** -c @CMAKE_INSTALL_FULL_SYSCONFDIR@/mbusd/mbusd-%i.conf -p /dev/%i

change the **-L -** to the directory and file that you would like the log file to be:

E.g.

ExecStart=@CMAKE_INSTALL_FULL_BINDIR@/mbusd -d -v2 **-L /var/log/mbusd/mbusd.log** -c @CMAKE_INSTALL_FULL_SYSCONFDIR@/mbusd/mbusd-%i.conf -p /dev/%i

zmata, 2018-04-04:
Hi, I've already changed -L - to -L /var/log/mbusd/%i before creating my deb package.
Thanks

closed 2018-04-04 by zmata