3cky/mbusd #33: high byte/word first to low byte/word first conversion

zmata, 2018-04-26:
Would it be possible to implement conversion options for:
High Byte First <--> Low Byte First
and 
High Word First <--> Low Word First
Thanks

3cky, 2018-04-27:
I think it will be little tricky because mbusd as gateway can be used to access multiple slaves in RTU network, so mentioned conversion options should be configured on a per-slave basis, but currently mbusd has no per-slave configuration options. Also I think the best place to do such kind of conversion is a TCP master accessing mbusd.

zmata, 2018-04-27:
I agree, but Loxone Smart Home miniserver only support it for reading registers, but not for writing.
I asked them to improve it, but they don't want to support it ;-(
We don't need it per slave. it's enough to have it per serial port (config file).