3cky/mbusd #38: mbusd getting hanged behind Ubuntu loaded on VirtualBox ( VM )

abhishekricky, 2018-09-27:
Hi,

I installed successfully now, it works intialy good for 5-10 seconds and then program mbusd needs to be restarted again 

Not sure what the issue , though there is only one master polling. I see following messages continuously 

7 Sep 2018 18:59:00 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:05 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:10 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:15 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:20 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:25 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:30 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:35 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:40 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:45 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:50 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 18:59:55 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 19:00:00 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 19:00:03 conn[10.0.2.2]: timeout, closing connection
27 Sep 2018 19:00:03 conn_close(): closing connection from 10.0.2.2
27 Sep 2018 19:00:05 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 19:00:07 conn[10.0.2.2]: timeout, closing connection
27 Sep 2018 19:00:07 conn_close(): closing connection from 10.0.2.2
27 Sep 2018 19:00:10 conn_open(): accepting connection from 10.0.2.2
27 Sep 2018 19:00:11 conn[10.0.2.2]: timeout, closing connection
27 Sep 2018 19:00:11 conn_close(): closing connection from 10.0.2.2

abhishekricky, 2018-09-27:
And ports are being shared using NAT option of Virtual Box

3cky, 2018-10-12:
Hello @abhishekricky 

Could you post additional info about your setup - versions of the host/guest OS and VirtualBox you're using? Thanks.

closed 2019-08-07 by stale[bot]