3cky/mbusd #44: debug option to see modbus info from serial device

shawnggraham, 2019-02-04:
Is there any way to log what mbusd is retrieving via /dev/ttyUSB0 before it is encapsulating into tcpip.  Also because im kind of new to this, how would my scada system need to communicate with a device that only understood rtu?  Would i need to create a device that read rtu over tcp or does mbusd treat it as actual modbus tcp?  The configuration i have is ... ignition scada system ->vpn-concentrator ->celular lte modem->raspbeerypie vpn client->raspberrypie-mbusd->/dev/ttyUSB0 ->waterlog-350xl(waterlogger). I can trace packets at each of the vpn end points and everything seems to be going ok but i cannot see what mbusd is getting and sending (not at the mbusd device anyway)  If that makes any sence.

3cky, 2019-02-05:
Did you try to increase log level verbosity? mbusd should dump received data to log file when its log level is set to 5 or higher value, so you could try to run it with `-v5` option.

nickma82, 2019-05-26:
please close this ticket too

closed 2019-10-23 by stale[bot]