3cky/mbusd #45: RTU over TCP?

nzfarmer1, 2019-02-09:
Hi,

Can I use your tool to map Modbus TCP to Modbus RTU over an IP network?

I'm using an Ethernet to Serial adaptor so can use socat to create a virtual serial port over IP, but the packet format is still RTU and socat is a little unstable through resets.

I was looking for something that proxy Modbus TCP to RTU but use an IP network (TCP or UDP)

3cky, 2019-02-11:
Well, since mbusd acts as a gateway translating Modbus TCP master requests to Modbus RTU RS-232/485 slaves network and back, I guess - yes, you can. :)

nzfarmer1, 2019-02-11:
But we'd have to adapt the code (which may be a reasonable project).

It appears that using the UDP option in our RTU to Ethernet adaptor gives us stability - fingers crossed.

socat -d -d -d pty,link=${VPORT},b${BAUD},echo=0,raw,unlink-close=0 udp:${USRIOTADDR}:${USRIOTPORT},reuseaddr

nickma82, 2019-05-26:
@nzfarmer1  Please close the ticket

closed 2019-05-30 by nzfarmer1