3cky/mbusd #53: Request to set address

vanrein, 2019-07-13:
This looks like a really nice, small and highly usable utility.  Compliments!

To my surprise, there is no way to set an IP address, is that correct?  At the very least localhost would be interesting.

Also, I would appreciate IPv6 support, though in reality I would probably add an access-constraining gateway that could limit World access.  I'd then run this daemon on localhost only.

Although I doubt you'll need is, here is some code similar to your socket module that I wrote elsewhere:
https://gitlab.com/arpa2/kip/blob/initial/src/socket.c
You're welcome to use (any part of) this, of course.

3cky, 2019-07-15:
You're right, there is no way to specify an IP address to bind for now, though mbusd socket-related code has support for binding to addresses different from `INADDR_ANY`. Add an option to bind to specified address shouldn't be a big deal, at least in IPv4 case. I'll look into it.

vanrein, 2019-07-15:
Wonderful, thanks!

3cky, 2019-07-17:
@vanrein I've added a config option 'address' and command-line option '-A' for setting up an IP address to bind. Both IPv4 and IPv6 are should be supported now, please test.

vanrein, 2019-07-17:
Hi 3cky,

It works like a charm, thank you very much!

I tested IPv4 and IPv6, both existing and non-existing addresses, and also the localhost addresses 127.0.0.1 and ::1.  It all worked as expected.

Good work :)
 -Rick

3cky, 2019-07-18:
Great, thanks for testing! I'll close this issue, feel free to open new issue if needed.

closed 2019-07-18 by 3cky