3cky/mbusd #55: Abnormal function on RPI

jeffersonpimenta, 2019-07-29:
I'm having problems trying to running mbusd on RPI with ScadaBR. Here my setup:

[image]

I made one mod on the PCB that converts RS485 to serial to suporto 3.3V, it works, but sometimes i have problems.

[image]

I'm have two of those making the conection between Rpi and energy meter

[image]

Without them, the conection works nice, but with them i have the following messages 

Jul 28 23:38:30 raspberrypi mbusd[954]: 28 Jul 2019 23:38:30 conn_open(): accepting connection from 127.0.0.1
Jul 28 23:38:30 raspberrypi mbusd[954]: 28 Jul 2019 23:38:30 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:30 raspberrypi mbusd[954]: 28 Jul 2019 23:38:30 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:30 raspberrypi mbusd[954]: 28 Jul 2019 23:38:30 tty: error in read() (port closed)
Jul 28 23:38:30 raspberrypi mbusd[954]: 28 Jul 2019 23:38:30 tty: error in read() (port closed)
Jul 28 23:38:31 raspberrypi mbusd[954]: 28 Jul 2019 23:38:31 tty: error in read() (port closed)
Jul 28 23:38:33 raspberrypi mbusd[954]: 28 Jul 2019 23:38:33 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:33 raspberrypi mbusd[954]: 28 Jul 2019 23:38:33 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:33 raspberrypi mbusd[954]: 28 Jul 2019 23:38:33 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:33 raspberrypi mbusd[954]: 28 Jul 2019 23:38:33 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:36 raspberrypi mbusd[954]: 28 Jul 2019 23:38:36 tty: error in read() (port closed)
Jul 28 23:38:36 raspberrypi mbusd[954]: 28 Jul 2019 23:38:36 tty: error in read() (port closed)
Jul 28 23:38:36 raspberrypi mbusd[954]: 28 Jul 2019 23:38:36 tty: error in read() (port closed)
Jul 28 23:38:39 raspberrypi mbusd[954]: 28 Jul 2019 23:38:39 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:39 raspberrypi mbusd[954]: 28 Jul 2019 23:38:39 tty: error in read() (port closed)
Jul 28 23:38:39 raspberrypi mbusd[954]: 28 Jul 2019 23:38:39 tty: error in read() (port closed)
Jul 28 23:38:41 raspberrypi mbusd[954]: 28 Jul 2019 23:38:41 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:41 raspberrypi mbusd[954]: 28 Jul 2019 23:38:41 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:41 raspberrypi mbusd[954]: 28 Jul 2019 23:38:41 tty: error in read() (port closed)
Jul 28 23:38:41 raspberrypi mbusd[954]: 28 Jul 2019 23:38:41 tty: error in read() (port closed)
Jul 28 23:38:41 raspberrypi mbusd[954]: 28 Jul 2019 23:38:41 tty: error in read() (port closed)
Jul 28 23:38:44 raspberrypi mbusd[954]: 28 Jul 2019 23:38:44 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:44 raspberrypi mbusd[954]: 28 Jul 2019 23:38:44 tty: error in read() (port closed)
Jul 28 23:38:44 raspberrypi mbusd[954]: 28 Jul 2019 23:38:44 tty: error in read() (port closed)
Jul 28 23:38:47 raspberrypi mbusd[954]: 28 Jul 2019 23:38:47 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:47 raspberrypi mbusd[954]: 28 Jul 2019 23:38:47 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:47 raspberrypi mbusd[954]: 28 Jul 2019 23:38:47 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:47 raspberrypi mbusd[954]: 28 Jul 2019 23:38:47 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:47 raspberrypi mbusd[954]: 28 Jul 2019 23:38:47 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:49 raspberrypi mbusd[954]: 28 Jul 2019 23:38:49 tty: error in read() (port closed)
Jul 28 23:38:49 raspberrypi mbusd[954]: 28 Jul 2019 23:38:49 tty_reopen(): can't open tty device /dev/ttyS0 (Input/output error)
Jul 28 23:38:49 raspberrypi mbusd[954]: 28 Jul 2019 23:38:49 tty: error in read() (port closed)
Jul 28 23:38:52 raspberrypi mbusd[954]: 28 Jul 2019 23:38:52 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:52 raspberrypi mbusd[954]: 28 Jul 2019 23:38:52 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:52 raspberrypi mbusd[954]: 28 Jul 2019 23:38:52 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:52 raspberrypi mbusd[954]: 28 Jul 2019 23:38:52 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:55 raspberrypi mbusd[954]: 28 Jul 2019 23:38:55 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:55 raspberrypi mbusd[954]: 28 Jul 2019 23:38:55 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:55 raspberrypi mbusd[954]: 28 Jul 2019 23:38:55 tty: error in read() (Resource temporarily unavailable)
Jul 28 23:38:57 raspberrypi mbusd[954]: 28 Jul 2019 23:38:57 tty: error in read() (port closed)
Jul 28 23:38:57 raspberrypi mbusd[954]: 28 Jul 2019 23:38:57 tty: error in read() (port closed)
Jul 28 23:38:57 raspberrypi mbusd[954]: 28 Jul 2019 23:38:57 tty: error in read() (port closed)

With USB dongle showed on the first photo, the system works, but not flawless. I have thoughts that mbusd doesn't read config file correctly

My conf file is:

device = /dev/ttyS0
speed = 9600
mode = 8n1
trx_control = rts
address = 0.0.0.0
port = 502
maxconn = 32
timeout = 60
retries = 0
pause = 2000
wait = 2000

Thanks!

3cky, 2019-07-31:
Hi @jeffersonpimenta,

First of all, make sure you have disabled serial console on /dev/ttyS0 (see an example [here](https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/#Disabling_the_Console)). 

Also, you have `trx_control = rts` in your config. Since RPi ttyS0 serial port doesn't have the RTS line, you probably would to replace `rts` with `addc` . If your RS232/485 adapter doesn't support ADDC mode (automatic data direction control), you could try to control it with GPIO pin using sysfs method.

jeffersonpimenta, 2019-07-31:
Now it works fine: i notice tx blink one extra time before request from modbus, it should be activity on terminal spit out in serial.

Note: rts and addc makes no diference for me, works on both mode. In thesis this board has automatic tx and rx flow control.

Anyway, thanks!

closed 2019-08-07 by jeffersonpimenta