3cky/mbusd #56: Service Failed to start

thelamnguyen72, 2020-04-12:
Hi Victor
I'm trying to use mbusd on my raspberry Pi as a Modbus TCP 2 serial server and every time I reboot the pi it always displayed a message : "Failed to start Modbus TCP to Modbus (RS-232/485) gateway.
Here is output form systemctl

```
root@raspberrypi:/home/pi# systemctl status mbusd@ttyUSB0.service
● mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-04-12 20:40:31 BST; 3min 7s ago
  Process: 317 ExecStart=/usr/local/bin/mbusd -d -v2 -L - -c /usr/local/etc/mbusd/mbusd-ttyUSB0.conf -p /dev/ttyUSB0 (code=exited, status=255/EXCEPTION)
 Main PID: 317 (code=exited, status=255/EXCEPTION)

Apr 12 20:40:31 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Service RestartSec=1s expired, scheduling restart.
Apr 12 20:40:31 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Scheduled restart job, restart counter is at 5.
Apr 12 20:40:31 raspberrypi systemd[1]: Stopped Modbus TCP to Modbus RTU (RS-232/485) gateway..
Apr 12 20:40:31 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Start request repeated too quickly.
Apr 12 20:40:31 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Failed with result 'exit-code'.
Apr 12 20:40:31 raspberrypi systemd[1]: Failed to start Modbus TCP to Modbus RTU (RS-232/485) gateway..
```
I've noticed the messages saying that: 
"Start request repeated too quickly"
"Failed with result 'exit-code'

Please advice on this matter. Thanks

3cky, 2020-04-14:
Did it work when started manually from the command line? If not, please start it with `-v9 -L/tmp/mbusd.log` options and post generated mbusd.log file here.

thelamnguyen72, 2020-04-14:
> CAUTION. This email originated from outside of the organization. Do not click links or open attachments unless you can confirm the sender and know the content is safe.

Hello Victor,

I'd like to say thank you for taking time to answer my query, appreciate for that. Per you question, the mbusd service failed to start at boot time as console displayed messages. To double check I run:

sudo systemctl status mbusd@ttyUSB0.service
and get this:
  mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled; vendor preset: e
   Active: failed (Result: exit-code) since Tue 2020-04-14 14:58:57 BST; 5h 51mi
  Process: 349 ExecStart=/usr/local/bin/mbusd -d -v2 -L - -c /usr/local/etc/mbus
 Main PID: 349 (code=exited, status=255/EXCEPTION)

Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Service RestartSe
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Scheduled restart
Apr 14 14:58:57 raspberrypi systemd[1]: Stopped Modbus TCP to Modbus RTU (RS-232
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Start request rep
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Failed with resul
Apr 14 14:58:57 raspberrypi systemd[1]: Failed to start Modbus TCP to Modbus RTU
so I run:
sudo systemctl stop mbusd@ttyUSB0.service
sudo systemctl start mbusd@ttyUSB0.service
sudo systemctl status mbusd@ttyUSB0.service
then got this
● mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled; vendor preset: e
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-04-14 20
  Process: 541 ExecStart=/usr/local/bin/mbusd -d -v2 -L - -c /usr/local/etc/mbus
 Main PID: 541 (code=exited, status=255/EXCEPTION)
which indicated that it's working.

then I run this

sudo mbusd -L/tmp/mbusd.log /dev/ttyUSB0 -s 38400 -P 502 -d -v9

but here is the log

14 Apr 2020 21:33:29 mbusd-0.4.1 started...
14 Apr 2020 21:33:29 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:33:29 conn_init() failed, exiting...
14 Apr 2020 21:34:10 mbusd-0.4.1 started...
14 Apr 2020 21:34:10 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:34:10 conn_init() failed, exiting...
14 Apr 2020 21:36:23 mbusd-0.4.1 started...
14 Apr 2020 21:36:23 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:36:23 conn_init() failed, exiting...
14 Apr 2020 21:36:31 mbusd-0.4.1 started...
14 Apr 2020 21:36:31 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:36:31 conn_init() failed, exiting...
14 Apr 2020 21:39:32 mbusd-0.4.1 started...
14 Apr 2020 21:39:32 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:39:32 conn_init() failed, exiting...
Here is my config file

#############################################
#                                           #
#    Sample configuration file for mbusd    #
#                                           #
#############################################

########## Serial port settings #############

# Serial port device name
device = /dev/ttyUSB0

# Serial port speed
speed = 38400

# Serial port mode
mode = 8n1

# RS-485 data direction control type (addc, rts, sysfs_0, sysfs_1)
trx_control = rts

# Sysfs file to use to control data direction
# trx_sysfile =

############# TCP port settings #############

# TCP server address to bind
address = 192.168.1.238

# TCP server port number
port = 502

# Maximum number of simultaneous TCP connections
maxconn = 32

# Connection timeout value in seconds
timeout = 60

######### Request/response settings #########

# Maximum number of request retries
retries = 3

# Pause between requests in milliseconds
pause = 250

# Response wait time in milliseconds
wait = 1000

One thing I noticed is my setting for serial port is ttyUSB0 but the log file showed /dev/ttyS0

Please advice. Thanks and have a great day.

Best Regards,

TLN

This email is intended solely for the person or entity to which it is addressed and may contain confidential and/or privileged information. Copying, forwarding or distributing this message and its attachments by persons or entities other than the intended recipient is prohibited. If you have received this email in error, please notify the sender immediately and delete the material from your computer and other electronic storage/delivery systems.

thelamnguyen72, 2020-04-14:
Hello Victor,

I'd like to say thank you for taking time to answer my query, appreciate for that. Per you question, the mbusd service failed to start at boot time as console displayed messages. To double check I run:

sudo systemctl status mbusd@ttyUSB0.service
and get this:
  mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled; vendor preset: e
   Active: failed (Result: exit-code) since Tue 2020-04-14 14:58:57 BST; 5h 51mi
  Process: 349 ExecStart=/usr/local/bin/mbusd -d -v2 -L - -c /usr/local/etc/mbus
 Main PID: 349 (code=exited, status=255/EXCEPTION)

Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Service RestartSe
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Scheduled restart
Apr 14 14:58:57 raspberrypi systemd[1]: Stopped Modbus TCP to Modbus RTU (RS-232
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Start request rep
Apr 14 14:58:57 raspberrypi systemd[1]: mbusd@ttyUSB0.service: Failed with resul
Apr 14 14:58:57 raspberrypi systemd[1]: Failed to start Modbus TCP to Modbus RTU
so I run:
sudo systemctl stop mbusd@ttyUSB0.service
sudo systemctl start mbusd@ttyUSB0.service
sudo systemctl status mbusd@ttyUSB0.service
then got this
● mbusd@ttyUSB0.service - Modbus TCP to Modbus RTU (RS-232/485) gateway.
   Loaded: loaded (/lib/systemd/system/mbusd@.service; enabled; vendor preset: e
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-04-14 20
  Process: 541 ExecStart=/usr/local/bin/mbusd -d -v2 -L - -c /usr/local/etc/mbus
 Main PID: 541 (code=exited, status=255/EXCEPTION)
which indicated that it's working.

then I run this

sudo mbusd -L/tmp/mbusd.log /dev/ttyUSB0 -s 38400 -P 502 -d -v9

but here is the log

14 Apr 2020 21:33:29 mbusd-0.4.1 started...
14 Apr 2020 21:33:29 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:33:29 conn_init() failed, exiting...
14 Apr 2020 21:34:10 mbusd-0.4.1 started...
14 Apr 2020 21:34:10 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:34:10 conn_init() failed, exiting...
14 Apr 2020 21:36:23 mbusd-0.4.1 started...
14 Apr 2020 21:36:23 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:36:23 conn_init() failed, exiting...
14 Apr 2020 21:36:31 mbusd-0.4.1 started...
14 Apr 2020 21:36:31 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:36:31 conn_init() failed, exiting...
14 Apr 2020 21:39:32 mbusd-0.4.1 started...
14 Apr 2020 21:39:32 conn_init(): can't open tty device /dev/ttyS0 (No such file or directory)
14 Apr 2020 21:39:32 conn_init() failed, exiting...
Here is my config file

#############################################
#                                           #
#    Sample configuration file for mbusd    #
#                                           #
#############################################

########## Serial port settings #############

# Serial port device name
device = /dev/ttyUSB0

# Serial port speed
speed = 38400

# Serial port mode
mode = 8n1

# RS-485 data direction control type (addc, rts, sysfs_0, sysfs_1)
trx_control = rts

# Sysfs file to use to control data direction
# trx_sysfile =

############# TCP port settings #############

# TCP server address to bind
address = 192.168.1.238

# TCP server port number
port = 502

# Maximum number of simultaneous TCP connections
maxconn = 32

# Connection timeout value in seconds
timeout = 60

######### Request/response settings #########

# Maximum number of request retries
retries = 3

# Pause between requests in milliseconds
pause = 250

# Response wait time in milliseconds
wait = 1000

One thing I noticed is my setting for serial port is ttyUSB0 but the log file showed /dev/ttyS0

Please advice. Thanks and have a great day.

Best Regards,

TLN


3cky, 2020-04-15:
> sudo mbusd -L/tmp/mbusd.log /dev/ttyUSB0 -s 38400 -P 502 -d -v9

Looks like you forgot to add `-p` option before device name `/dev/ttyUSB0`, so mbusd is trying to open wrong serial port.

thelamnguyen72, 2020-04-15:
Hello Victor,

Yes, you're right. my command line was missing -p option.
After manually start the service here is the log output

15 Apr 2020 16:51:38 mbusd-0.4.1 started...
15 Apr 2020 16:51:38 tty: state now is TTY_PAUSE
15 Apr 2020 16:51:38 conn_loop(): select(): max_sd = 4, t_out = 000000:100000
15 Apr 2020 16:51:38 conn_loop(): select() returns 0
15 Apr 2020 16:51:38 tty: state now is TTY_READY
15 Apr 2020 16:51:38 conn_loop(): select(): max_sd = 4, t_out = 000060:000000
15 Apr 2020 16:52:16 conn_loop(): select() returns -1
15 Apr 2020 16:52:16 Terminated by signal: SIGINT
As a side note, I'm running your software on raspberry Pi model B.
Is there any hardware constrains that caused this problem with this platform? Thank you for your time.

Best Regards,
TLN

3cky, 2020-04-17:
Well, the log output looks normal. Did you try to connect to mbusd with a TCP client? If mbusd works when started by hand, I would suspect some problem in your systemd instance config file, like wrong bind address.

thelamnguyen72, 2020-04-17:
Hello Victor,

Thank you again for your support. I’ve tried connecting the device in the scheme listed below:

[Host PC running ModScan software] <-------> cat5 cable <-------> [Raspberry PI running Mbusd]<-------> [usb2ser converter] <----->[rs232/rs485 converter]<-----> [Real modbus slave device ID:3]

I tried to poll the slave device without success. I know the slave device is working because I can poll it from the host PC via serial link without any problem as shown here:

[Host PC running ModScan software] <-------->  [usb2ser converter] <----->[rs232/rs485 converter]<-----> [Real modbus slave device ID:3]

Should I set the IP address of the Pi and my host PC to use static IP addresses?

Best Regards,

TheLam Nguyen
Design Engineer III - Electrical, Subsea Control Systems
Dril-Quip, Inc.
6401 N. Eldridge Pkwy
Houston, Tx 77041
Tel: (713) 939-0047 ext 6488
E-mail: Thelam_Nguyen@dril-quip.com<mailto:Thelam_Nguyen@dril-quip.com>

closed 2020-09-15 by stale[bot]