3cky/mbusd #57: Modbus RTU over TCP

embeddman, 2020-05-12:
Hi! Are you planning to add modbus RTU over TCP feature? For example, instead of using Modbus TCP packet, deploy Modbus RTU over TCP packet. It's very comfortable thing for getting info via virtual COM port (emulating Modbus RTU device using TCP connection).

3cky, 2020-05-15:
Though it's an interesting idea, I'm still not sure it will fit into the mbusd concepts of multiplexed access to the RTU network. Need to look into this more thoroughly.

dasrue, 2023-08-13:
I have an initial implementation of this [here](https://github.com/dasrue/mbusd). If this would be useful to anyone else I will create a pull request, but I will do some more testing first

kpr0th, 2023-12-22:
@dasrue -- if I'm reading correctly, your implementation is still a ModbusTCP server (slave), but you added an option to make a ModbusRTU "encapsulated in TCP packets" client (master) connection to a remote device instead of accessing the remote device over a serial (TTY) connection?

I.e.:  [client-device] ---ModbusTCP(502)---> [mbusd] ---ModbusRTU-encapsulated-in-TCP-packets---> [remote device]?

If so, that's a pretty cool option.  I'd encourage submitting as a PR.

closed 2026-09-20 by embeddman

schweigstill, 2026-09-21:
There are many devices in the field which traditionally had an EIA-485 for Modbus RTU. Then someone replaced the  EIA-485 transceiver by a "stupid" serial to TCP converter without properly converting it to Modbus TCP. The biggest problem is that T35 (three and a half UART character timeout) has to be handled on the remote side of the TCP connection, but without knowing the UART/EIA-485/RTU baudrate.

So I fully agree that this shouldn't be supported by a protocol stack/implementation. Instead either use a normal EIA-485 for RTU or use a proper converter/gateway.