3cky/mbusd #59: Delay after sending frame via tty

embeddman, 2020-05-21:
Hi!
Faced with trouble when sending frames via tty with parity enabled. RTS changes with gpio via sysfs. The solution was in extending delay after sending frame otherwise RTS changes every time on last byte and corrupt it. Maybe delay was calculated not taking into account that parity add extra bit to frame?

3cky, 2020-05-21:
Hi @embeddman, сould you provide some info about the hardware you're using?

embeddman, 2020-05-22:
Hi @3cky, I'm using imx6ul from NXP.

3cky, 2020-05-22:
@embeddman I've pushed some fixes related to the timing calculations, please test.

embeddman, 2020-05-22:
> @embeddman I've pushed some fixes related to the timing calculations, please test.

@3cky Now works fine, even more accurate timing than my solution.
[image]

closed 2020-05-23 by embeddman