3cky/mbusd #70: Raspberry Pi ttyUSB0 errors

zaricd, 2020-11-24:
Hi 

When I run mbusd with ttyUSB0 on my Raspberry Pi Compute Module I get those errors:

24 Nov 2020 13:43:37 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
24 Nov 2020 13:43:38 tty: error in read() (Resource temporarily unavailable)
24 Nov 2020 13:43:38 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
24 Nov 2020 13:43:38 tty: error in read() (Bad address)

I tested ttyUSB0 from Linux terminal programs and it is communicating with my computer both way.

I run mbusd from command line :  sudo mbusd -d -t -L - -v2 -p /dev/ttyUSB0 -s 9600 -m 8N1

I found where error comes , from conn_read in conn.c but not sure why, it looks something connected to buffers.
Any hint where I can start from to troubleshoot this ?

3cky, 2020-11-28:
Hello @zaricd, could you run mbusd with `-v9` option and post the complete log here?

zaricd, 2020-11-28:
Victor

Thanks for replay, I went and debug it in CLion, I found that problem was with echo coming back from my RS485 hardware, when in sending mode. Disabling echo solved problem.

closed 2020-11-28 by zaricd