3cky/mbusd #72: Tty response timing out

b0czek, 2021-03-01:
Hello. I wanted to use your gateway with a motor driver on raspberry pi, but I have encountered a problem, that the tty response always times out which would suggest that the driver does not even respond to requests. When I try to connect with it via RCU on Windows machine it works flawlessly. I use the same transmission parameters as on the Windows machine and I tried to tinker with timeout and wait times but it didn't help either. The one thing that caught my attention was 'Station No.' which is, i suppose, slave selection and it was set to 1. I don't really know much about RS-485 and I haven't really looked into the code but I wonder if this might be a problem. I've attached logs of the program where i try to connect. [mbusd.log](https://github.com/3cky/mbusd/files/6061054/mbusd.log)

3cky, 2021-03-09:
Looks like your TCP master sends requests to the RTU device with Slave ID set to `FF`:

`01 Mar 2021 11:03:30 conn[192.168.1.53]: request: [ff][03][03][00][00][01]`

Make sure your RTU slave has the same Unit/Slave ID. If not, change its value either on master or slave device.

b0czek, 2021-03-11:
Thank you for responding, I don't have the hardware at hand anymore unfortunately, so I can't verify that but it seems that it is the case. Thank you for help.

closed 2021-03-11 by b0czek

hACKrus, 2023-09-24:
Hello.
I have similar problem. Slave ID is 31. Program tries to connect but gives timeout error. mbus.conf have only changed setting to 8n2.

# docker run -ti -v /dev/ttyRS485-1:/dev/serial --privileged -v /etc/mbusd.conf:/opt/mbusd.conf -p 1502:502  --name=mbusd  quenorha/mbusd -v9

24 Sep 2023 16:10:07 mbusd-0.5.1 started...
24 Sep 2023 16:10:07 tty: trying to open /dev/ttyS0 (speed 9600 mode 8n2)
24 Sep 2023 16:10:07 tty: state now is TTY_PAUSE
24 Sep 2023 16:10:07 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:07 conn_loop(): select() returns 0
24 Sep 2023 16:10:07 tty: state now is TTY_READY
24 Sep 2023 16:10:07 conn_loop(): select(): max_sd = 4, t_out = 4227156:000060
24 Sep 2023 16:10:13 conn_loop(): select() returns 1
24 Sep 2023 16:10:13 conn_open(): accepting connection from 192.168.0.135
24 Sep 2023 16:10:13 queue_new_elem(): length now is 1
24 Sep 2023 16:10:13 conn[192.168.0.135]: state now is CONN_HEADER
24 Sep 2023 16:10:13 conn_loop(): select(): max_sd = 5, t_out = 4227156:000060
24 Sep 2023 16:10:14 conn_loop(): select() returns 1
24 Sep 2023 16:10:14 conn[192.168.0.135]: state now is CONN_RQST_FUNC
24 Sep 2023 16:10:14 conn_loop(): select(): max_sd = 5, t_out = 4227156:000060
24 Sep 2023 16:10:14 conn_loop(): select() returns 1
24 Sep 2023 16:10:14 conn[192.168.0.135]: read request fc 3
24 Sep 2023 16:10:14 conn[192.168.0.135]: state now is CONN_RQST_TAIL
24 Sep 2023 16:10:14 conn[192.168.0.135]: request: [1f][03][00][00][00][02]
24 Sep 2023 16:10:14 conn[192.168.0.135]: state now is CONN_TTY
24 Sep 2023 16:10:14 tty: state now is TTY_RQST
24 Sep 2023 16:10:14 conn_loop(): select(): max_sd = 4, t_out = 4227156:000060
24 Sep 2023 16:10:14 conn_loop(): select() returns 1
24 Sep 2023 16:10:14 tty: written 8 bytes
24 Sep 2023 16:10:14 tty: request written (total 8 bytes)
24 Sep 2023 16:10:14 tty: state now is TTY_RESP
24 Sep 2023 16:10:14 tty: estimated 9 bytes, waiting 519478 usec
24 Sep 2023 16:10:14 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:14 conn_loop(): select() returns 0
24 Sep 2023 16:10:14 tty: response timeout
24 Sep 2023 16:10:14 tty: attempt to retry request (1 of 3)
24 Sep 2023 16:10:14 tty: state now is TTY_RQST
24 Sep 2023 16:10:14 tty: written 8 bytes
24 Sep 2023 16:10:14 tty: request written (total 8 bytes)
24 Sep 2023 16:10:14 tty: state now is TTY_RESP
24 Sep 2023 16:10:14 tty: estimated 9 bytes, waiting 519478 usec
24 Sep 2023 16:10:14 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:15 conn_loop(): select() returns 0
24 Sep 2023 16:10:15 tty: response timeout
24 Sep 2023 16:10:15 tty: attempt to retry request (2 of 3)
24 Sep 2023 16:10:15 tty: state now is TTY_RQST
24 Sep 2023 16:10:15 tty: written 8 bytes
24 Sep 2023 16:10:15 tty: request written (total 8 bytes)
24 Sep 2023 16:10:15 tty: state now is TTY_RESP
24 Sep 2023 16:10:15 tty: estimated 9 bytes, waiting 519478 usec
24 Sep 2023 16:10:15 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:16 conn_loop(): select() returns 0
24 Sep 2023 16:10:16 tty: response timeout
24 Sep 2023 16:10:16 tty: attempt to retry request (3 of 3)
24 Sep 2023 16:10:16 tty: state now is TTY_RQST
24 Sep 2023 16:10:16 tty: written 8 bytes
24 Sep 2023 16:10:16 tty: request written (total 8 bytes)
24 Sep 2023 16:10:16 tty: state now is TTY_RESP
24 Sep 2023 16:10:16 tty: estimated 9 bytes, waiting 519478 usec
24 Sep 2023 16:10:16 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:16 conn_loop(): select() returns 0
24 Sep 2023 16:10:16 tty: response timeout
24 Sep 2023 16:10:16 conn[192.168.0.135]: state now is CONN_RESP
24 Sep 2023 16:10:16 tty: state now is TTY_PAUSE
24 Sep 2023 16:10:16 conn_loop(): select(): max_sd = 5, t_out = 4227156:000000
24 Sep 2023 16:10:16 conn_loop(): select() returns 1
24 Sep 2023 16:10:16 conn[192.168.0.135]: state now is CONN_HEADER
24 Sep 2023 16:10:16 conn_loop(): select(): max_sd = 5, t_out = 4227156:000000
24 Sep 2023 16:10:16 conn_loop(): select() returns 1
24 Sep 2023 16:10:16 conn[192.168.0.135]: state now is CONN_RQST_FUNC
[... 407 lines cut ...]
24 Sep 2023 16:10:33 tty: attempt to retry request (3 of 3)
24 Sep 2023 16:10:33 tty: state now is TTY_RQST
24 Sep 2023 16:10:33 tty: written 8 bytes
24 Sep 2023 16:10:33 tty: request written (total 8 bytes)
24 Sep 2023 16:10:33 tty: state now is TTY_RESP
24 Sep 2023 16:10:33 tty: estimated 9 bytes, waiting 519478 usec
24 Sep 2023 16:10:33 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:34 conn_loop(): select() returns 0
24 Sep 2023 16:10:34 tty: response timeout
24 Sep 2023 16:10:34 conn[192.168.0.135]: state now is CONN_RESP
24 Sep 2023 16:10:34 tty: state now is TTY_PAUSE
24 Sep 2023 16:10:34 conn_loop(): select(): max_sd = 5, t_out = 4227156:000000
24 Sep 2023 16:10:34 conn_loop(): select() returns 1
24 Sep 2023 16:10:34 conn_close(): closing connection from 192.168.0.135
24 Sep 2023 16:10:34 queue_delete_elem(): length now is 0
24 Sep 2023 16:10:34 conn_loop(): select(): max_sd = 4, t_out = 4227156:000000
24 Sep 2023 16:10:34 conn_loop(): select() returns 0
24 Sep 2023 16:10:34 tty: state now is TTY_READY
24 Sep 2023 16:10:34 conn_loop(): select(): max_sd = 4, t_out = 4227156:000060
^C24 Sep 2023 16:10:36 conn_loop(): select() returns -1
24 Sep 2023 16:10:36 conn_loop(): error in select() (Invalid argument)
24 Sep 2023 16:10:36 conn_open(): error in accept() (Resource temporarily unavailable)
24 Sep 2023 16:10:36 tty: read(): no data available
24 Sep 2023 16:10:36 Terminated by signal: SIGINT
`