3cky/mbusd #74: tty: dropped 205 bytes

towen, 2021-03-25:
Hi,

When running on a Raspberry Pi connected to a Modbus RTU slave over RS232 (TTL) I get no response from mbusd.
I am using the 'sunspec2' library. If I connect sunspec2 as an RTU client directly to the serial port (ttyUSB0) it works fine. But if I connect mbusd to the serial port and try to connect sunspec2 as a TCP client to mbusd, it times out.
If I monitor the serial port, I can see the data returned, but mbusd never sees it.
Setting loglevel 9, I can see that it says the tty dropped some bytes - I guess that's the problem

```
$ mbusd -d -v 9 -p /dev/ttyUSB0 -s 115200 -P 5502
25 Mar 2021 17:13:11 mbusd-0.5.0 started...
25 Mar 2021 17:13:11 tty: trying to open /dev/ttyUSB0 (speed 115200 mode 8N1)
25 Mar 2021 17:13:23 tty: state now is TTY_PAUSE
25 Mar 2021 17:13:23 conn_loop(): select(): max_sd = 4, t_out = 000000:100000 
25 Mar 2021 17:13:23 conn_loop(): select() returns 0 
25 Mar 2021 17:13:23 tty: state now is TTY_READY
25 Mar 2021 17:13:23 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:25 conn_loop(): select() returns 0 
25 Mar 2021 17:13:25 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:27 conn_loop(): select() returns 0 
25 Mar 2021 17:13:27 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:29 conn_loop(): select() returns 1 
25 Mar 2021 17:13:29 conn_open(): accepting connection from 127.0.0.1
25 Mar 2021 17:13:29 queue_new_elem(): length now is 1
25 Mar 2021 17:13:29 conn[127.0.0.1]: state now is CONN_HEADER
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 5, t_out = 000002:000000 
25 Mar 2021 17:13:29 conn_loop(): select() returns 1 
25 Mar 2021 17:13:29 conn[127.0.0.1]: state now is CONN_RQST_FUNC
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 5, t_out = 000002:000000 
25 Mar 2021 17:13:29 conn_loop(): select() returns 1 
25 Mar 2021 17:13:29 conn[127.0.0.1]: read request fc 3
25 Mar 2021 17:13:29 conn[127.0.0.1]: state now is CONN_RQST_TAIL
25 Mar 2021 17:13:29 conn[127.0.0.1]: request: [00][03][9c][40][00][64]
25 Mar 2021 17:13:29 conn[127.0.0.1]: state now is CONN_TTY
25 Mar 2021 17:13:29 tty: state now is TTY_RQST
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:29 conn_loop(): select() returns 1 
25 Mar 2021 17:13:29 tty: written 8 bytes
25 Mar 2021 17:13:29 tty: request written (total 8 bytes)
25 Mar 2021 17:13:29 conn[127.0.0.1]: state now is CONN_HEADER
25 Mar 2021 17:13:29 tty: state now is TTY_PAUSE
25 Mar 2021 17:13:29 conn[127.0.0.1]: broadcast request sent
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 5, t_out = 000000:100000 
25 Mar 2021 17:13:29 conn_loop(): select() returns 1 
25 Mar 2021 17:13:29 tty: dropped 205 bytes
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 5, t_out = 000000:061069 
25 Mar 2021 17:13:29 conn_loop(): select() returns 0 
25 Mar 2021 17:13:29 tty: state now is TTY_READY
25 Mar 2021 17:13:29 conn_loop(): select(): max_sd = 5, t_out = 000002:000000 
25 Mar 2021 17:13:31 conn_loop(): select() returns 1 
25 Mar 2021 17:13:31 conn[127.0.0.1]: timeout, closing connection
25 Mar 2021 17:13:31 conn_close(): closing connection from 127.0.0.1
25 Mar 2021 17:13:31 queue_delete_elem(): length now is 0
25 Mar 2021 17:13:31 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:33 conn_loop(): select() returns 0 
25 Mar 2021 17:13:33 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:35 conn_loop(): select() returns 0 
25 Mar 2021 17:13:35 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:37 conn_loop(): select() returns 0 
25 Mar 2021 17:13:37 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:39 conn_loop(): select() returns 0 
25 Mar 2021 17:13:39 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:41 conn_loop(): select() returns 0 
25 Mar 2021 17:13:41 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
25 Mar 2021 17:13:43 conn_loop(): select() returns 0 
25 Mar 2021 17:13:43 conn_loop(): select(): max_sd = 4, t_out = 000002:000000 
^C25 Mar 2021 17:13:44 conn_loop(): select() returns -1 
25 Mar 2021 17:13:44 Terminated by signal: SIGINT
```

3cky, 2021-03-30:
>> 25 Mar 2021 17:13:29 conn[127.0.0.1]: request: [00][03][9c][40][00][64]
>> 25 Mar 2021 17:13:29 conn[127.0.0.1]: broadcast request sent

Looks like your RTU slave is configured to use address 0 which is reserved for a broadcast address in Modbus protocol. In broadcast mode, the master sends a request to all slaves and no response is returned to broadcast requests sent by the master. Hence mbusd ignores this unexpected response message and logs the message about dropped bytes.

Could you configure your RTU device to use another address, like 1? It should fix the problem.

towen, 2021-03-30:
Aha of course! Thanks! :)

closed 2021-03-30 by towen