3cky/mbusd #80: service stops working after several hours - looking for help

octane321, 2021-12-23:
Hi there and first of all a very big THX for this wonderful software.

I´m running the mbusd on a rpi 3b with a iobroker slave. I´m reading my solar charge controllers wich are connected via RS485 and it worked just fine for month until i had to replace the SD-Card and reinstall alsmost everything.
On the new OS the service works for hours and then connectionproblems occur.
It can only be solved by restarting the service with 
`sudo systemctl status mbusd@ttyXR_EP0.service`

The Logfile is very big >100MB until the error occures, so i try to paste relevant lines only (hope so)

```
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_HEADER
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 5, t_out = 000000:099128 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_RQST_FUNC
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 5, t_out = 000000:098355 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 conn[127.0.0.1]: read request fc 3
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_RQST_TAIL
22 Dec 2021 16:39:16 conn[127.0.0.1]: request: [03][03][90][14][00][01]
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_TTY
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000000:097506 
22 Dec 2021 16:39:16 conn_loop(): select() returns 0 
22 Dec 2021 16:39:16 tty: state now is TTY_RQST
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000059:000000 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 tty: written 8 bytes
22 Dec 2021 16:39:16 tty: request written (total 8 bytes)
22 Dec 2021 16:39:16 tty: state now is TTY_RESP
22 Dec 2021 16:39:16 tty: estimated 7 bytes, waiting 501301 usec
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000000:501301 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 tty: read 7 bytes
22 Dec 2021 16:39:16 tty: rx offset is 0
22 Dec 2021 16:39:16 tty: read 7 bytes of 7, offset 0
22 Dec 2021 16:39:16 tty: state now is TTY_PROC
22 Dec 2021 16:39:16 tty: response read (total 7 bytes, offset 0 bytes)
22 Dec 2021 16:39:16 tty: response is correct
22 Dec 2021 16:39:16 tty: response: [02][04][02][00][00][fd][30]
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_RESP
22 Dec 2021 16:39:16 tty: state now is TTY_PAUSE
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 6, t_out = 000000:100000 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_HEADER
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 6, t_out = 000000:099701 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_RQST_FUNC
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 6, t_out = 000000:098162 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 conn[127.0.0.1]: read request fc 3
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_RQST_TAIL
22 Dec 2021 16:39:16 conn[127.0.0.1]: request: [02][03][91][07][00][01]
22 Dec 2021 16:39:16 conn[127.0.0.1]: state now is CONN_TTY
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000000:097590 
22 Dec 2021 16:39:16 conn_loop(): select() returns 0 
22 Dec 2021 16:39:16 tty: state now is TTY_RQST
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
22 Dec 2021 16:39:16 conn_loop(): select() returns 1 
22 Dec 2021 16:39:16 tty: written 8 bytes
22 Dec 2021 16:39:16 tty: request written (total 8 bytes)
22 Dec 2021 16:39:16 tty: state now is TTY_RESP
22 Dec 2021 16:39:16 tty: estimated 7 bytes, waiting 501301 usec
22 Dec 2021 16:39:16 conn_loop(): select(): max_sd = 4, t_out = 000000:501301 
22 Dec 2021 16:39:17 conn_loop(): select() returns 1 
22 Dec 2021 16:39:17 tty: read 7 bytes
22 Dec 2021 16:39:17 tty: rx offset is 0
22 Dec 2021 16:39:17 tty: read 7 bytes of 7, offset 0
22 Dec 2021 16:39:17 tty: state now is TTY_PROC
22 Dec 2021 16:39:17 tty: response read (total 7 bytes, offset 0 bytes)
22 Dec 2021 16:39:17 tty: response is correct
22 Dec 2021 16:39:17 tty: response: [03][03][02][16][10][ce][28]
[... 99 lines cut ...]
23 Dec 2021 02:57:50 tty: response timeout
23 Dec 2021 02:57:50 conn[127.0.0.1]: state now is CONN_RESP
23 Dec 2021 02:57:50 tty: state now is TTY_PAUSE
23 Dec 2021 02:57:50 conn_loop(): select(): max_sd = 5, t_out = 000000:100000 
23 Dec 2021 02:57:50 conn_loop(): select() returns 1 
23 Dec 2021 02:57:50 conn[127.0.0.1]: state now is CONN_HEADER
23 Dec 2021 02:57:50 conn_loop(): select(): max_sd = 5, t_out = 000000:099726 
23 Dec 2021 02:57:50 conn_loop(): select() returns 0 
23 Dec 2021 02:57:50 tty: state now is TTY_READY
23 Dec 2021 02:57:50 conn_loop(): select(): max_sd = 5, t_out = 000059:000000 
23 Dec 2021 02:57:53 conn_loop(): select() returns 1 
23 Dec 2021 02:57:53 conn_close(): closing connection from 127.0.0.1
23 Dec 2021 02:57:53 queue_delete_elem(): length now is 0
23 Dec 2021 02:57:53 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
23 Dec 2021 02:58:53 conn_loop(): select() returns 0 
23 Dec 2021 02:58:53 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
23 Dec 2021 02:59:53 conn_loop(): select() returns 0 
23 Dec 2021 02:59:53 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
23 Dec 2021 03:00:53 conn_loop(): select() returns 0 
23 Dec 2021 03:00:53 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
23 Dec 2021 03:01:54 conn_loop(): select() returns 0 
23 Dec 2021 03:01:54 conn_loop(): select(): max_sd = 4, t_out = 000060:000000 
```

thanks!

3cky, 2021-12-23:
Hello @octane321! 

Could you provide some additional info about your setup? Which OS version, iobroker and serial adapter hardware did you run successfully before, and what is changed after?

octane321, 2021-12-23:
Hi @3cky 
thanx for your quick reply
OS: Raspbian 11 (bullseye)
Node.js v14.18.2
NPM 6.14.15
js-controller v3.3.21

The serial adapter is this one https://www.ebay.de/itm/142659997743
The hardware didn´t change - its the same as before

Only new SD-Card and fresh OS install.

THX!

3cky, 2021-12-24:
> The serial adapter is this one [https://www.ebay.de/itm/142659997743](https://bre.is/redirect/https://www.ebay.de/itm/142659997743)

It's FT232RL based, right?

> sudo systemctl status mbusd@ttyXR_EP0.service

AFAIK, FT232RL linux device name should be `ttyUSB0`. So why service references `ttyXR_EP0` device name?

octane321, 2021-12-24:
Sorry, i should have mentioned this.
i have more than one of these USB-RS485 converters connected and to identify the right one, i had to add a udev rule depending on the USB Port where the adapter is connected.

so this is output of /ls -lah /dev/tty*

[image]

i also tried with /dev/ttyUSB1 but it does not make a difference

3cky, 2021-12-27:
Well, that explains a lot. :)

Had a look at the log file. It seems mbusd receives TCP request for single discrete input number `0x2000` read (function code `0x02`) from slave device with address `0x02`:

> 23 Dec 2021 02:57:48 conn[127.0.0.1]: request: [02][02][20][00][00][01]

and sends it to RTU device:

> 23 Dec 2021 02:57:48 tty: request written (total 8 bytes)

but doesn't receive any response after 3 retries:

> ...
> 23 Dec 2021 02:57:50 tty: attempt to retry request (3 of 3)
> 23 Dec 2021 02:57:50 tty: state now is TTY_RQST
> 23 Dec 2021 02:57:50 tty: written 8 bytes
> 23 Dec 2021 02:57:50 tty: request written [...]

Not quite sure why this is happening. Maybe the USB adapter is silently disconnecting? Do you have any messages related to the USB subsystem in dmesg log?

octane321, 2022-01-01:
You are absolutely right:
Found following message with `dmesg -T`

`[Sa Jan  1 03:07:00 2022] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[Sa Jan  1 04:04:57 2022] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32`

i read something about it and maybe the new kernel has a bug or something....do you have any idea how i could fix the issue?

THX

octane321, 2022-01-04:
Issue is solved - it was a bad USB Kable!
Working perfect again!
@3cky Thanks again for your help

closed 2022-01-04 by octane321