3cky/mbusd #84: Crash due to missing `logw()` argument

vanrein, 2022-03-15:
I triggered a simple oversight in `master`, in `src/conn.c` it says

```
#ifdef DEBUG
    logw(5, "conn[%s]: request data len changed from %d to %d",
         MB_FRAME(conn->buf, MB_LENGTH_L), len);
#endif
```

There should be 3 arguments after the `logw()` format string, not 2.  The crashing stopped when I changed `[%s]` to `[%%s]`.

I may have been doing something wrong —I am starting a project and am may be sending invalid packets— but your dæmon might want to stick around to tell me off :-)

dgoo2308, 2022-04-06:
@vanrein 

please try with adding `conn->remote_addr` in the logw of `conn_fix_request_header_len`

```c
void
conn_fix_request_header_len(conn_t *conn, unsigned char len)
{
  if (MB_FRAME(conn->buf, MB_LENGTH_L) != len)
  {
#ifdef DEBUG
    logw(5, "conn[%s]: request data len changed from %d to %d",
         conn->remote_addr,MB_FRAME(conn->buf, MB_LENGTH_L), len);
#endif
    MB_FRAME(conn->buf, MB_LENGTH_L) = len;
  }
}
```

vanrein, 2022-04-06:
That fixes the problem, indeed.  Thanks.

To test, I reconstructed the situation because it was originally my bug triggering yours; my bug has since been fixed.  So I added an `else` branch printing to also produce this error, I saw that crash in the same manner, and then applied your fix to see the crash gone.

3cky, 2022-04-06:
Thanks, guys. Indeed `conn->remote_addr` argument was missing. Will fix it.

commit 06ccb293 (2022-04-06): Fix crash due to missing logw() argument (fixes #84)
closed 2022-04-06 by 3cky