3cky/mbusd #87: mbusd

ensmartcontrols, 2022-10-19:
Hi,
  I deployed mbusd application in Tinker board R2.0 ,Tested with Normal Modbus Devices , It is communicating with well,

Now we are going to integrate Danfoss VFD using mbusd application, We got Modbus addressing as below. Whether mbusd will support on this.

Model: Danfoss Fc102
(Six Digit Holding Register)
415010- Running Hours
415020- Kwh Counter 
etc.,
[DanfossFC102 (1).xls](https://github.com/3cky/mbusd/files/9816550/DanfossFC102.1.xls)
Herewith I attached Danfoss VFD Modbus Addressing Details.

Thanks & Regards,

K. L. Narayanan
India

3cky, 2022-10-28:
Hello,

Since "six-digit" is just a form of representation for Modbus entities, mbusd should work OK with it if your Modbus TCP software is doing its part of the translation from 6-digit to standard function/register form.

ensmartcontrols, 2023-04-08:
Hi,

It is communicating, thanks for your reply, It is delayed due to site commissioning, We completed Integration successfully, Thanks 

Regards,

K.L.Narayanan