3cky/mbusd #88: Missing device is retried forever [and blocks anything else]

vanrein, 2022-10-25:
This is some output from v0.5.1 without the expected device connected:

```
DEBUG: Modbus TCP send = 00 00 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP send = 00 01 00 00 00 06 03 06 00 35 00 01 -> #12
DEBUG: Modbus TCP send = 00 02 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP recv = 00 00 00 00 00 03 03 83 0b -> #9
DEBUG: Modbus TCP send = 00 03 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP recv = 00 00 00 00 00 03 03 83 0b -> #9
DEBUG: Modbus TCP send = 00 04 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP send = 00 05 00 00 00 06 03 06 00 35 00 00 -> #12
DEBUG: Modbus TCP recv = 00 00 00 00 00 03 03 83 0b -> #9
DEBUG: Modbus TCP send = 00 06 00 00 00 06 03 03 00 32 00 04 -> #12
```

Note that the transaction identifier in responses is stuck at 00 00 and nothing else is tried.

Are you intentionally sending a provisional response "hold on, will keep trying... forever if need be"?

In any case, with other devices also on the bus, their communication also gets halted, so I suspect this is a bug, namely forgetting to remove the command that just failed.

vanrein, 2022-10-25:
It's very instructive to remove the power from one device while scanning multiple.

vanrein, 2022-10-25:
This is an example where I rip the power from one device and the other seems to end up queued forever while `mbusd` blocks on the powerless one and continues to try to send transaction 000e.  Not ideal from a stability viewpoint :-)

```
Connected.  Me Happy.
Entering loop.  Should exit when mbusd disconnects.
DEBUG: Modbus TCP send = 00 00 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP send = 00 01 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 00 00 00 00 0b 03 03 08 00 00 00 15 00 12 00 01 -> #17
Device temperatures: 0 21 18 1
DEBUG: Modbus TCP recv = 00 01 00 00 00 07 10 04 04 00 d4 02 5a -> #13
Temperature 21.200000, moist 60.200000%
DEBUG: Modbus TCP send = 00 02 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 02 00 00 00 07 10 04 04 00 d4 02 5a -> #13
Temperature 21.200000, moist 60.200000%
DEBUG: Modbus TCP send = 00 03 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 03 00 00 00 07 10 04 04 00 d4 02 5a -> #13
Temperature 21.200000, moist 60.200000%
DEBUG: Modbus TCP send = 00 04 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 04 00 00 00 07 10 04 04 00 d4 02 5a -> #13
Temperature 21.200000, moist 60.200000%
DEBUG: Modbus TCP send = 00 05 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 05 00 00 00 07 10 04 04 00 d4 02 59 -> #13
Temperature 21.200000, moist 60.100000%
DEBUG: Modbus TCP send = 00 06 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 07 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP recv = 00 06 00 00 00 07 10 04 04 00 d4 02 5a -> #13
Temperature 21.200000, moist 60.200000%
DEBUG: Modbus TCP recv = 00 07 00 00 00 0b 03 03 08 00 00 00 15 00 12 00 01 -> #17
Device temperatures: 0 21 18 1
DEBUG: Modbus TCP send = 00 08 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 08 00 00 00 07 10 04 04 00 d3 02 59 -> #13
Temperature 21.100000, moist 60.100000%
DEBUG: Modbus TCP send = 00 09 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 09 00 00 00 07 10 04 04 00 d4 02 59 -> #13
Temperature 21.200000, moist 60.100000%
DEBUG: Modbus TCP send = 00 0a 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0a 00 00 00 07 10 04 04 00 d4 02 59 -> #13
Temperature 21.200000, moist 60.100000%
DEBUG: Modbus TCP send = 00 0b 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0b 00 00 00 07 10 04 04 00 d4 02 59 -> #13
Temperature 21.200000, moist 60.100000%
DEBUG: Modbus TCP send = 00 0c 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 0d 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP recv = 00 0c 00 00 00 07 10 04 04 00 d4 02 59 -> #13
Temperature 21.200000, moist 60.100000%
DEBUG: Modbus TCP recv = 00 0d 00 00 00 0b 03 03 08 00 00 00 15 00 12 00 01 -> #17
Device temperatures: 0 21 18 1
DEBUG: Modbus TCP send = 00 0e 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 0f 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 10 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP send = 00 11 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 12 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 13 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP send = 00 14 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 15 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP send = 00 16 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 17 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP send = 00 18 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP send = 00 19 00 00 00 06 03 03 00 32 00 04 -> #12
DEBUG: Modbus TCP send = 00 1a 00 00 00 06 10 04 00 01 00 02 -> #12
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP recv = 00 0e 00 00 00 03 10 84 0b -> #9
DEBUG: Modbus TCP send = 00 1b 00 00 00 06 10 04 00 01 00 02 -> #12
```

vanrein, 2022-10-25:
In case you find it useful, this is the [generating code](https://gitlab.com/groengemak/mbusc/) for the output above.

vanrein, 2022-10-25:
Interestingly, the return of power does not recover the interaction.  The state of `mbusd` appears disturbed, and it cranks out only those failures for the once-failed transaction.

3cky, 2022-10-28:
Could you also provide the mbusd debug log file, as described [here](https://github.com/3cky/mbusd#reporting-bugs)?

vanrein, 2022-11-16:
Sorry for the slow response; I was distracted by IETF 115.

I cannot reproduce it at this point, even though I tried two flavours of hardware (USB to RS-485) tokens.

I made many changes, but not to this core, so this leaves me a bit dissatisfied.  Alas, I suppose we should close for now, and perhaps I can reopen the issue if I run into this again.  I will then also supply the mbusd debugging output.

closed 2022-11-16 by vanrein