3cky/mbusd #92: /dev/ttyS0?

FrazKK, 2023-03-31:
Hi guys, I'm sure this is very straight forward for many of you but please help me out, I've been searching everywhere for a couple of days now and beginning to pull my hair out!
I'm attempting to setup the sunsynk addon on home assistant on an Orange Pi 3 lts, I've installed the modbus TCP to modbus RTU gateway. Under device it says /Dev/ttys0, I have read should be ttyUSB0, is my orange pi not picking up the USB to RS485 converter at all? 
Can anyone point me in a direction to some instructions for what I've got going wrong please?

kinkazu, 2023-03-31:
Unplug USB/RS485 converter, restart PI, plug converter in, type dmesg in console and check if USB converter has been picked up by the system and if yes, you should see it’s tty name.

FrazKK, 2023-04-01:
Thank you for your reply Kinkazu. It say ttyACM0, does that make sense? I was expecting ttyUSB0

kinkazu, 2023-04-01:
If that is the name of tty that dmesg shows, that is the one.

FrazKK, 2023-04-03:
Thank you Kinkazu, appreciate the help!

falcon35180, 2023-09-17:
As I understand, USB UARTs that use their own custom protocol, such as FTDI, show up as ttyUSBx devices. It appears that your converter has a UART that implements the USB CDC protocol, so it appears as ttyACMx.

closed 2025-01-10 by 3cky