3cky/mbusd #94: CRC Enable/Disable

ensmartcontrols, 2023-04-12:
One of our site We used mbusd Modbus gateway(Using Tinker Board ) application for integrating Energy meters.
We faced "Message Timeout Error",  But we were getting values using USB to serial -Windows PC using Modscan. Same Energy meter loop is working with Trane SC+ controller and Field Server gateways.

We observed at site our customer was laid Modbus Signal cable along with Energy meter power cable(230 V AC), Then we laid separate cable from our Tinker board to Energy meters , it was communicating.

We assume mbusd Application is Implemented with CRC . Due to improper cabling it was not communicating. 

We couldn't able to convince customer, because they are not technical to understand, they are stating it is communicating of USB to serial with Modscan and our application is wrong.

Modbus Message Timeout is because of CRC ?, if yes so, is it possible to enable/disable CRC to prove us.

3cky, 2023-04-13:
Could you provide a [log file](https://github.com/3cky/mbusd#reporting-bugs) for problematic devices? I'll take a look.

ensmartcontrols, 2023-04-16:
We spent lot of time for evaluating the issue and identifies it is because of cable,customer is not willing to accept it. At present we don't have access that site for testing, 
To solve issue they installed Field server gateway and completed the integration.
 
Note:
They haven't changed the cabling for Energy meter.

At same site we have done integration(using Asus tinker board - Mbusd application) of other system like UPS, Water leak detection, VFD all are working fine. 
If we get access at same site or any other site integration will update you with error log.

kpr0th, 2023-12-25:
Compiling in debug mode would have shown the issue clearly.  Looking at the code in conn.c starting around line 460, if a read fails the crc check, it only shows this detail if #DEBUG.  Recommend adding a logw(2...) call if not defined #debug, so that any future people running into crc issues can see it clearly just by adding "-d" to the command line params.