3cky/mbusd #100: Error when building on Raspberry Pi

Roselt, 2023-12-26:
I get the following error when running the make command:

Scanning dependencies of target mbusd
[  8%] Building C object CMakeFiles/mbusd.dir/src/main.c.o
/home/pi/mbusd.git/src/main.c: In function ‘main’:
/home/pi/mbusd.git/src/main.c:263:9: error: a label can only be part of a statement and a declaration is not a statement
  263 |         char *logfilenamevalue = strdup(optarg);
      |         ^~~~
/home/pi/mbusd.git/src/main.c:264:9: error: expected expression before ‘char’
  264 |         char *logfilename = util_trim(logfilenamevalue);
      |         ^~~~
/home/pi/mbusd.git/src/main.c:265:21: error: ‘logfilename’ undeclared (first use in this function); did you mean ‘logfullname’?
  265 |         if (!strlen(logfilename))
      |                     ^~~~~~~~~~~
      |                     logfullname
/home/pi/mbusd.git/src/main.c:265:21: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/mbusd.dir/build.make:82: CMakeFiles/mbusd.dir/src/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:123: CMakeFiles/mbusd.dir/all] Error 2
make: *** [Makefile:182: all] Error 2

3cky, 2023-12-27:
Thanks for reporting! Should be fixed by 5b59f5b, please test.

Roselt, 2023-12-27:
Thank you. Tested and working.

closed 2023-12-27 by Roselt