3cky/mbusd #102: Docker-compose.yaml file ?

championc, 2024-02-19:
Can I make the installation using a docker-compose.yaml ?

Can you share a sample ?

jpmeijers, 2024-02-19:
```
services:
  mbusd:
    restart: unless-stopped
    privileged: true
    build:
      context: https://github.com/jpmeijers/mbusd.git
      dockerfile: Dockerfile
    volumes:
      - /dev:/dev
      - ./mbusd.conf:/etc/mbusd.conf
    ports:
      - 502:502
```

championc, 2024-02-19:
Many thanks

A question - for the volumes, what is the difference to set between 
/dev:/dev
and maybe
/dev/ttyUSB0:/dev/ttyUSB0

jpmeijers, 2024-02-20:
The docker-compose file mounts the whole `/dev` directory into the container, so it's up to your mbusd.conf file to specify the correct serial port to use. You can change the compose file to only mount a single tty by doing something like `/dev/ttyUSB0:/dev/ttyUSB0`, but then you need to specify the serial port in two locations.

If you do a `ls /dev/tty*` on the host, you will see which ones are available. In your case it looks like you do not have a ttyUS0.

My setup has two USB-Serial devices, so both a ttyUSB0 and a ttyUSB1. These numbers can be assigned at random, so I used Udev rules to create additional tty devices called `ttyRS485` and `ttyRS232`. I then link in the RS485 one into this container for mbusd, keeping the default ttyUSB0 name inside the container.
`/dev/ttyRS485:/dev/ttyUSB0`

championc, 2024-02-20:
Got it all sorted in the end

Many thanks

closed 2024-02-20 by championc