3cky/mbusd #108: exec format error when trying to start container on PI-OS 12 (64 bit)

JuergenOrth, 2024-06-27:
Any hint why it doesn't work?

root@pi51:~/docker/mbusd# grep VERSION= /etc/os-release
VERSION="12 (bookworm)"

root@pi51:~/docker/mbusd# cat *l
version: "3"
services:
  mbusd:
    restart: always
    image: 3cky/mbusd
    volumes:
      - /etc/mbusd.conf:/etc/mbusd.conf
      - /dev:/dev
    ports:
      - 502:502

root@pi51:~/docker/mbusd# docker-compose up
Creating network "mbusd_default" with the default driver
Pulling mbusd (3cky/mbusd:)...
latest: Pulling from 3cky/mbusd
bca4290a9639: Already exists
b6b6114c4de7: Pull complete
Digest: sha256:dae23e3e9ef546850dc9eb0e9bc8e1d11676748866744dbe2df44c11a8fafc8e
Status: Downloaded newer image for 3cky/mbusd:latest
Creating mbusd_mbusd_1 ... done
Attaching to mbusd_mbusd_1
mbusd_1  | exec /usr/bin/mbusd: exec format error

root@pi51:~/docker/mbusd# docker inspect 3cky/mbusd | jq '.[0].Architecture'
"arm64"

root@pi51:~/docker/mbusd#

G33KatWork, 2024-08-01:
I am facing the same issue on ARMv7.

The executable in the container is built for x86_64. That can't work:

```
[andy@energypi ~]$ docker cp elastic_pike:/usr/bin/mbusd .
Successfully copied 66kB to /home/andy/.

[andy@energypi ~]$ file mbusd 
mbusd: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-x86_64.so.1, BuildID[sha1]=06c71f037094333c794c7878d528772561896c45, with debug_info, not stripped
```

I haven't looked into how the container is actually built, so I don't know why this happens.

I can't even get the docker container to build myself without stripping the `--platform=$BUILDPLATFORM` in the first `FROM`. I am too dumb to supply that argument it seems.

However, after removing that from the Dockerfile and then running `docker build -t 3cky/mbusd:latest .` on the ARM machine itself, I got the container to work. That's at least a solution until the build for other architectures than x86 is fixed.

3cky, 2024-08-07:
I did some fixes to Dockerfile, please pull latest image and test.

closed 2025-01-10 by 3cky