3cky/mbusd #110: .deb installation for aarch64

rubenvanwanzeele, 2024-08-06:
Hi there,

I would need a .deb installation file for aarch64 for mbusd, but I currently fail to create one. Can you guide me on how this can be done?

zmata, 2024-08-06:
Hi, you need to use checkinstall command

rubenvanwanzeele, 2024-08-06:
@zmata that worked, thx, but still struggling to resolve the issue on your repo...

3cky, 2024-08-07:
> Hi there,
> 
> I would need a .deb installation file for aarch64 for mbusd, but I currently fail to create one. Can you guide me on how this can be done?

On your aarch64 machine, try to execute the following commands in `mbusd` sources directory:
```
mkdir ./build
cd ./build
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
make all
cpack -G DEB
```
After that, the generated .deb file should be placed in `build` directory.

closed 2024-08-08 by rubenvanwanzeele