3cky/mbusd #112: Cannot get ttyUSB0 working

searlecom, 2024-09-25:
I have a Waveshare Industrial USB to RS485 Converter which is on ttyUSB0 on my Pi5

I have got it working in Windows with some software called ModbusPoll and can read from my Sunsynk inverter.

I cannot seem to get mbusd to read the serial port at all.  I get the below in the log:

25 Sep 2024 22:49:58 mbusd-0.5.3 started...

25 Sep 2024 22:49:58 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8n1)

I have used the below to create the docker container:

version: '3'
services:
  mbusd:
    container_name: mbusd
    group_add:
      - dialout
    user: root
    privileged: true
    ports:
      - 502:502
    volumes:
      - /dev:/dev
      - /opt/mbusd:/etc
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    image: 3cky/mbusd:latest 

Has anyone seen this before?

I have also tried connecting with Putty but not sure it that will work at all.

mStirner, 2024-09-25:
What does "I cannot seem to get mbusd to read the serial port at all" mean?

- Did you try to use mbusd directly wihtout docker?
- What command did you try to talk to mbusd?
- Is the adapter correctly regocnized by the host OS?
- How did you test it?

Never had any issue with the cheapest USB/RS485 convertes & mbusd.
Please add some more details and answer the questions above to get help.

searlecom, 2024-09-26:
I have just installed it on the host OS still getting the same error, I ran it as root.

It looks like the device is good in the host OS, details are below:

udevadm info /dev/ttyUSB0
P: /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1:1.0/ttyUSB0/tty/t>
M: ttyUSB0
R: 0
U: tty
D: c 188:0
N: ttyUSB0
L: 0
S: serial/by-id/usb-FTDI_FT232R_USB_UART_A10OGNJP-if00-port0
S: serial/by-path/platform-xhci-hcd.0-usb-0:1:1.0-port0
E: DEVPATH=/devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1:1.0/ttyUS>
E: DEVNAME=/dev/ttyUSB0
E: MAJOR=188
E: MINOR=0
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=5906231
E: ID_BUS=usb
E: ID_MODEL=FT232R_USB_UART
E: ID_MODEL_ENC=FT232R\x20USB\x20UART
E: ID_MODEL_ID=6001
E: ID_SERIAL=FTDI_FT232R_USB_UART_A10OGNJP
E: ID_SERIAL_SHORT=A10OGNJP
E: ID_VENDOR=FTDI
E: ID_VENDOR_ENC=FTDI
E: ID_VENDOR_ID=0403
E: ID_REVISION=0600
E: ID_TYPE=generic
E: ID_USB_MODEL=FT232R_USB_UART
E: ID_USB_MODEL_ENC=FT232R\x20USB\x20UART
E: ID_USB_MODEL_ID=6001
E: ID_USB_SERIAL=FTDI_FT232R_USB_UART_A10OGNJP
E: ID_USB_SERIAL_SHORT=A10OGNJP
E: ID_USB_VENDOR=FTDI
E: ID_USB_VENDOR_ENC=FTDI
E: ID_USB_VENDOR_ID=0403
E: ID_USB_REVISION=0600
E: ID_USB_TYPE=generic
E: ID_USB_INTERFACES=:ffffff:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=ftdi_sio
E: ID_VENDOR_FROM_DATABASE=Future Technology Devices International, Ltd
E: ID_MODEL_FROM_DATABASE=FT232 Serial (UART) IC
E: ID_PATH=platform-xhci-hcd.0-usb-0:1:1.0
E: ID_PATH_TAG=platform-xhci-hcd_0-usb-0_1_1_0
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10OGNJP-if00-port0 /dev/serial/by-path/pl>
E: TAGS=:systemd:
E: CURRENT_TAGS=:systems:

dmesg | grep ttyUSB0
[    3.227342] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0

Can you explain how you would test it, I am no Linux expert.

Thanks
Mike

mStirner, 2024-09-26:
> Can you explain how you would test it, I am no Linux expert.

mbusd wrapps the plain modbus RTU into a IP stack.

What would be in plain modbus RTU `\x01\x06\x00\x02\x00\x04` (example from my AC/climate unit) is wrapped/over mbusd `x00\x01\x00\x00\x00\x06\x01\x06\x00\x02\x00\x04`.

Complete example with mbusd & netcat: `echo -n -e '\x00\x01\x00\x00\x00\x06\x01\x06\x00\x02\x00\x04' | nc -q 1 climate.lan 502` This would set the fans speed on my AC to the value 4, on register 2. `'\x00\x01\x00\x00\x00\x06` is needed for the encapsulation in TCP/IP.

Thats why i asked, how you test it.
Its likely you missed/misformatted the modbus message.

The examples above needed to be adapted to your need.
But thats how i test nearly every device i use with mbusd.
Format the modbus message properly, and write it with `echo` & `netcat` to your destination.

flx25, 2024-10-17:
I have the Same issue with a Waveshare USB to RS485 Converter. Am running it under WSL where i can acess the adapter with other programms but mbusd gives me the following:

`17 Oct 2024 14:46:14 mbusd-0.5.3 started...
17 Oct 2024 14:46:14 tty: trying to open /dev/ttyUSB0 (speed 19200 mode 8E1)
17 Oct 2024 14:46:14 tty: trying to enable RS-485 support for /dev/ttyUSB0
17 Oct 2024 14:46:14 conn_init(): can't open tty device /dev/ttyUSB0 (Inappropriate ioctl for device)
17 Oct 2024 14:46:14 conn_init() failed, exiting...`

pflugj, 2024-10-17:
at flx25
Did you try to disable RS-485 support! ( I believe this is for COM-port devices)

closed 2025-01-10 by 3cky