3cky/mbusd #118: Add pre-build binariers

mStirner, 2024-12-08:
I love mbusd and use it nearly everywhere where i have to work with USB/RS485 Adapters and Modbus.
But what always bother is the fact, that you have to compile it from source everytime.

Is there a way/possibility to distirubte pre-build binaries?
E.g.:
- mbusd binary
- systemd unit file
- example.conf

You pick then what you need/want and install it where it belongs, or use the mbusd binary as standalone executable without the need for building.

mStirner, 2024-12-08:
Especial for low power devices like older raspberry pi's or other single board computers, it would be awesome to not have to install a full build tool chain.

Benik3, 2024-12-13:
You can try `make altinstall` and `dpkg-deb --build` to make your own .deb installer

mStirner, 2025-01-22:
How to build for RHEL distros like CentOS/Rocky?

Benik3, 2025-02-06:
I don't have experience with RPM packages, sorry.
But you can try to check CPackRPM.